madison-embedded/gcc-builds #3: Common: Missing md command

vkottler, 2017-06-15:
Need to implement `md <address> <count>` so we can display contents of memory in both raw hexadecimal and ASCII formats (both at once not either or).

This implementation should hopefully check if the memory access will be legal since not all of the virtual 4 GB space is mapped, but the alternative is to develop a robust BusFault Handler to recover from these accesses.

vkottler, 2017-06-19:
It's ready in [cmd_mem.c](https://github.com/madison-embedded/gcc-builds/blob/master/common/cmd_mem.c) just needs to be implemented.

vkottler, 2017-06-19:
[Datasheet](https://github.com/madison-embedded/documentation/blob/master/stm32/f7xx/datasheet.pdf) for memory map.

vkottler, 2017-06-19:
Example:

[image]

On the right side if the bytes were ASCII the corresponding characters would be displayed which is great for looking at stored strings.

vkottler, 2017-06-19:
ASCII example:

[image]

Tserendulam, 2017-06-22:
Sounds good !!

vkottler, 2017-07-05:
[image]
[image]

Looks like we might need to refactor this a bit to match endiannesses from the ELF object dump (accessible via `make dump`).

Also flash can be accessed from another address:

[image]

vkottler, 2017-07-05:
[image]

Probably should be `unsigned int` since our integers are 32 bits.

vkottler, 2017-07-28:
Wondering if this might be done?

chroro, 2017-07-30:
i think this was fixed