madison-embedded/gcc-builds #11: Common: missing reset command

vkottler, 2017-06-19:
Being able to reset the board from the command line would be a great utility.

vkottler, 2017-06-22:
I recommend cloning [this repo](https://github.com/madison-embedded/documentation) to get all of the necessary documentation.

The [STM32F767ZI reference manual](https://github.com/madison-embedded/documentation/blob/master/stm32/f7xx/RM.pdf) says the following:

[image]

When we consult the [ARM Cortex-M7 Reference Manual](https://github.com/madison-embedded/documentation/blob/master/ARM/cortex_m7.pdf) as instructed, surprise! There's no description for the "Application Interrupt and Reset Control Register".

When we search it [online here](http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.dui0646a/Cihehdge.html) we can actually read about it:

[image]

hmm... yeah I'm sure it's somewhere in these manuals!

davidguyuchen, 2017-06-26:
It seems we can take advantage of NVIC_SYSTEMRESET command documented in core_cm7.h file in the CMSIS files folder. It basically overwrites the reset register bits to implement a software reset, which is pretty much what we are looking for.

[image]

vkottler, 2017-06-26:
Oh dang, good find. There's always useful stuff in there! Feel free to make a command-line command that calls it and test it out

davidguyuchen, 2017-06-28:
Has been implemented and pushed.

closed 2017-06-28 by davidguyuchen