madison-embedded/gcc-builds #8: Common: Missing a command for toggling pins

vkottler, 2017-06-19:
This may be the single most useful command-line utility we can provide.

Something like `pin [<pin_num> | <PORT><PIN>] [on | off]` would work best where `pin_num` is a pin from the female headers and `<PORT><PIN>` is something like **A8** or **C5** (may also be routed to one of the female header pins).

Eventually we'll want to support aliases for pins such as braking_solenoid_X so a struct that also implements that capability would be ideal.

vkottler, 2017-06-19:
STM32 Nucleo 144 related: [I/O schematic](https://github.com/madison-embedded/documentation/blob/master/stm32/nucleo/ioSch.PNG), [Badgerloop I/O documentation](https://docs.google.com/spreadsheets/d/1tzqCBdWPqIOUyLG5kfgT8CnwwVi2Sx8zOyC4Jp1TQsU/edit?usp=sharing)

vkottler, 2017-06-21:
It might be better to support read and write, in the future we'll also need an analog read function.

vkottler, 2017-06-21:
Want to make sure to check if a pin is already enabled and serving as an alternate function pin

vkottler, 2017-06-22:
add info to: `pin [<pin_num> | <PORT><PIN>] [on | off | info]`

Come up with a way to print info for all pins or all pins from a certain port.

csgreen3, 2017-06-29:
Besides alternate functions and analog information this is completed

vkottler, 2017-06-29:
Have `pin <PORT> [info]` display any potential alias of each pin in parentheses at the end

csgreen3, 2017-06-30:
[image]

vkottler, 2017-06-30:
Make the boolean flag mean the alias should not be used to initialize the pin, allow user to have full contorl over aliased pins

csgreen3, 2017-07-16:
added initialization

closed 2017-07-16 by csgreen3