madison-embedded/gcc-builds #9: Common: missing memmap command

vkottler, 2017-06-19:
A command `memmap` should be implemented that displays where different physical hardware peripherals (i.e. flash, RAM, core registers, peripheral registers) are mapped to.

It's not necessary to map out each peripheral, but something like this would work well:

[image]

vkottler, 2017-06-19:
For 32-bit microprocessors, it won't be necessary to create a different virtual and physical column (but if that is somehow implemented definitely make note of it!).

Most things also won't be in the order of megabytes, this example comes from an NXP T1042E.

Microprocessors also likely won't have peripherals such as PCI-express or external DDR memory.

vkottler, 2017-06-19:
STM32F767ZI datasheet can be downloaded [here](https://github.com/madison-embedded/documentation/blob/master/stm32/f7xx/datasheet.pdf).

chroro, 2017-06-21:
Sounds good !! Ill start gathering information today.

vkottler, 2017-07-05:
Thanks for mapping all of the peripherals. I will be nice to be able to test drivers by manually writing bytes before having to develop the entire driver interface.

Do you think you could add entries at the beginning that map out the physical memory from a higher level of abstraction? I was thinking of things from page 106-110 on the datasheet.

Tserendulam, 2017-07-05:
You mean the boundaries of AHB1,  AHB2 and etc ? Or do you want something like the following 
**Bus      Peripheral       Address       Size 
AHB1   Some Device  0x000000     10000** 

The staff that I wrote is basically everything from pages 108 - 110.  I can add 107.

chroro, 2017-07-05:
Haha, I commented from Tserendulam's account by mistake.

vkottler, 2017-07-28:
Wondering if this might be done?

chroro, 2017-07-28:
Sure. I cant come today. But Ill be at the lab the rest of the weekend.

vkottler, 2017-08-08:
I'd like to use some of the linker script definitions to get some runtime memory usage information:
```
/* Linker script to place sections and symbol values. Should be used together
 * with other linker script that defines memory regions FLASH and RAM.
 * It references following symbols, which must be defined in code:
 *   Reset_Handler : Entry of reset handler
 * 
 * It defines following symbols, which code can use without definition:
 *   __exidx_start
 *   __exidx_end
 *   __copy_table_start__
 *   __copy_table_end__
 *   __zero_table_start__
 *   __zero_table_end__
 *   __etext
 *   __data_start__
 *   __preinit_array_start
 *   __preinit_array_end
 *   __init_array_start
 *   __init_array_end
 *   __fini_array_start
 *   __fini_array_end
 *   __data_end__
 *   __bss_start__
 *   __bss_end__
 *   __end__
 *   end
 *   __HeapLimit
 *   __StackLimit
 *   __StackTop
 *   __stack
 */
```

I'm only concerned mainly with `__data_start__`, `__data_end__`, `__bss_start__`, `__bss_end__`, `__HeapLimit`, `__StackLimit`, `__StackTop`, and `__stack`.

I've been getting runtime warnings from the LwIP stack about memory usage so hoping to find out more about what can be done