madison-embedded/gcc-builds #15: STM32 Nucleo 144 F767ZI: Missing ADC Driver

vkottler, 2017-06-27:
Needed for pressure sensor readings and thermistors.

vkottler, 2017-06-27:
[image]

Datasheet:
[image]
[image]
[image]

vkottler, 2017-06-29:
needs refactoring but is essentially done

vkottler, 2017-07-28:
Remember to change 6 bit accuracy, finish `ar` command

vkottler, 2017-07-28:
thermistor code from pod 1: [https://github.com/vkottler/badgerloopControls/blob/master/Pic32/peripherals/C/thermistor.c](https://github.com/vkottler/badgerloopControls/blob/master/Pic32/peripherals/C/thermistor.c)

closed 2017-08-03 by csgreen3

csgreen3, 2017-08-03:
can read and write to all adc ar prints out all ar alias or ar port pin will print out a specific one