madison-embedded/gcc-builds #17: Build: Incorporate Build Timestamp to Post output

vkottler, 2017-06-27:
It's possible to use a pre-processor macro to create a string in C that will be equal to the current time stamp but this requires that the file containing the macro is re-built each time. 

This is contingent on the build process, make will only re-compile files that have been touched or modified more recently than the latest build files (which can of course be over written with `make clean`).

It might be easiest to somehow use the Makefile to create some file, `timestamp.c` for instance for each build. I'm guessing there are many ways to skin this cat.

Here's an example of what I'm looking for:

[image]

csgreen3, 2017-07-29:
I was getting this warning so I ended up commenting it out.
 
common/post.c: In function 'print_post_info':
<command-line>:0:11: warning: passing argument 4 of 'strftime' makes pointer from integer without a cast
common/post.c:35:42: note: in expansion of macro '_GIT_TIME'
  strftime(buff, sizeof(_GIT_TIME), "%c", _GIT_TIME);
                                          ^
In file included from /usr/include/newlib/stdio.h:29:0,
                 from common/post.c:1:
/usr/include/newlib/time.h:61:11: note: expected 'const struct tm *' but argument is of type 'int'
 size_t    _EXFUN(strftime, (char *__restrict _s,
           ^

vkottler, 2017-08-02:
it's hard to get the actual build timestamp to work without generating a header file in a Makefile rule so I'm keeping it to git commit timestamp + committer and hash, plus major and minor version

the problem is if you don't modify the source file that prints post output you won't have an updated timestamp

closed 2017-08-02 by vkottler