madison-embedded/gcc-builds #19: Common: Missing float command

vkottler, 2017-06-27:
Different ARM cores have varying levels of hardware / dedicated FPU support.

We need a float command that implements the following argument pattern:

`float [ add | subtract | multiply | divide | + | - | * | / ] <arg1> <arg2>`

In addition to performing the calculation, it would be great to have some diagnostic information about how the operation was carried out, including amount of time / number of instructions required if possible.

vkottler, 2017-06-29:
Best if done before next weekend (:

vkottler, 2017-07-12:
Can you build this in Keil uVision IDE and compare performance?

davidguyuchen, 2017-07-12:
Yep, I will give it a try and I'm assuming we are still using the C standard library?

vkottler, 2017-07-12:
Yeah, there's a different way to go about re-targeting `printf` in Keil though

closed 2017-07-17 by vkottler