madison-embedded/gcc-builds #25: STM32 Nucleo 144 F767ZI: Finish Pin Aliases

vkottler, 2017-06-30:
Use the [STM32 Nucleo 144 Manual](https://github.com/madison-embedded/documentation/blob/master/stm32/nucleo/nuc144.pdf) to finish populating `GPIO_TABLE` in [`include/gpio_alias.h`](https://github.com/madison-embedded/gcc-builds/blob/master/include/gpio_alias.h):

Example:

[image]

vkottler, 2017-06-30:
Also include the JTAG pins for the STM32F767ZI as described by the [Reference Manual](https://github.com/madison-embedded/documentation/blob/master/stm32/f7xx/RM.pdf):

[image]

Also check the reset values of GPIOA and GPIOB to see if we missed any of those pins

vkottler, 2017-07-11:
Will work on this today now that I mapped many pins using STM32CubeMX.

vkottler, 2017-07-13:
Shouldn't take too long to create the pin-initialization function and populate the rest of the pins.

What will take the longest is getting the alternate function values.

vkottler, 2017-07-17:
Probably just need to look this over once more and grab a few free pins for PWM / EXTI and GPIO.

csgreen3, 2017-07-29:
Updated all the pins from shield spread sheet. May want to change the names.  I changed retros from input capture to EXTI but we have E11 and E 13 that are on the same interrupt line that we may want to change.  but other than that we should have the pins all picked out.