madison-embedded/gcc-builds #35: Device Driver: MPU9250 (I2C)

vkottler, 2017-07-28:
[Breakout board from SparkFun](https://www.sparkfun.com/products/13762).
- [Datasheet](https://cdn.sparkfun.com/assets/learn_tutorials/5/5/0/MPU9250REV1.0.pdf)
- [Register Map](https://cdn.sparkfun.com/assets/learn_tutorials/5/5/0/MPU-9250-Register-Map.pdf)
 - [GitHub library source](https://github.com/sparkfun/MPU-9250_Breakout/tree/master/Libraries/Arduino/src)

I would suggest beginning by converting the library code to work with just `C` instead of `CPP` and changing I2C read and write function names to work with ours

vkottler, 2017-07-28:
FYI here's a previous driver I had been working on: [https://github.com/vkottler/badgerloopControls/blob/master/Pic32/peripherals/C/MPU9250.c](https://github.com/vkottler/badgerloopControls/blob/master/Pic32/peripherals/C/MPU9250.c)

chroro, 2017-07-30:
Does your code from the previous project work ? 
If yes , we can just  modify it to match ours.

chroro, 2017-07-30:
We will probably just change the GitHub library source to match our project.

vkottler, 2017-07-31:
Yeah it works, that sounds like the best approach

vkottler, 2017-08-03:
```
uint8_t readByte(uint8_t address, uint8_t subAddress)
{
  
 uint8_t *buffer =  (uint8_t *) malloc(sizeof(uint8_t));
 *buffer = subAddress;
 I2C_setSlaveAddr(address,  MPU9250_config.I2C_BASE);
 I2C_Master_Transmit_Data( MPU9250_config.I2C_BASE, 1, buffer);
 I2C_Master_Receive_Data( MPU9250_config.I2C_BASE, 1, buffer);
 I2C_Read_Data_Ready( MPU9250_config.I2C_BASE);
  
 return *buffer; 

}
```

Isn't `uint8_t *buffer =  (uint8_t *) malloc(sizeof(uint8_t));` the exact same as `uint8_t buffer;` and then passing `&buffer` to the function?

Consider the significant overhead in making a memory allocation library call

chroro, 2017-08-05:
We were using **&buffer** before we noticed a bug in **writeByte** function. Interestingly, since we don't busy wait in writeByte function until data is transmitted, next call to write function overwrites the data in the stack. Since stack contains the data (&buffer ) to send via I2C next writeByte call may overwrite the data to transmit from the first writeByte call.  Therefore, we had to malloc memory, to not overwrite any data.

chroro, 2017-08-05:
We also did not busy wait in readBytes. So we decided to call malloc again. I believe in the end we decided to busy wait in readBytes function so that we will return with the proper value in the passed buffer argument. Since we are busy waiting in readBytes and readByte we can use &buffer. we can change that )) good catch.