madison-embedded/gcc-builds #45: STM32F767ZI: Investigate ADC_TARGET_FREQ

vkottler, 2017-08-11:
[Cooper's commit](https://github.com/madison-embedded/gcc-builds/commit/0ce12a448ec53f7e35b67704589e96c4512d696f) fixes an issue where `ar` command hangs after I committed code that changed the core clock frequencies.

Code that uses `ADC_TARGET_FREQ`:

```
static bool adc_init_clk(ADC_TypeDef *adc) {
	uint8_t divisor = 2;
	uint32_t mask = 0;

	/*enable clock in RCC register */
	switch ((uint32_t) adc) {
		case ADC1_BASE: mask = RCC_APB2ENR_ADC1EN; break;
		case ADC2_BASE: mask = RCC_APB2ENR_ADC2EN; break;
		case ADC3_BASE: mask = RCC_APB2ENR_ADC3EN; break;
	}
	RCC->APB2ENR |= mask;

	if(adcFreq == 0){
		while (divisor <= 8 && APB2_F / divisor > ADC_TARGET_FREQ)
			divisor += 2;
		if (divisor > 8) return false;
		adcFreq = APB2_F / divisor;
		ADC->CCR &= ~ADC_CCR_ADCPRE_Msk;
		switch (divisor) {
			case 4: ADC->CCR |= ADC_CCR_ADCPRE_0; break;
			case 6: ADC->CCR |= ADC_CCR_ADCPRE_1; break;
			case 8: ADC->CCR |= ADC_CCR_ADCPRE_1 | ADC_CCR_ADCPRE_0; break;
		}
	}

	adc_init_sampling(adc);

	return true;
}
```

vkottler, 2017-08-11:
[image]

csgreen3, 2017-08-15:
Should be fixed may need to look into sampling rate

closed 2017-08-15 by csgreen3